AutonomyLab/create_robot #13: Add ability to change mode

jacobperron, 2016-03-28:
Should add a subscriber that listens to a custom message for changing the mode of Create (OFF, PASSIVE, SAFE, and FULL).

jacobperron, 2016-06-20:
Added functionality to [set-modes](https://github.com/AutonomyLab/create_autonomy/tree/set-modes) branch. Will merge after testing.

andrewrgross, 2021-10-06:
@jacobperron I'm trying to send a Roomba a command to go into passive mode from ROS. How far did you get with this branch?

jacobperron, 2021-10-07:
@andrewrgross It was so long ago, I don't recall exactly why I stopped working on this. If I remember correctly, the branch is pretty much feature-complete, but I may have ran into a bug/behaviour issue that I didn't resolve.

matttsmith, 2022-02-15:
@andrewrgross Did you find anything? I'm going to start working on this as well - just started working with ROS on a Roomba Create 2.

andrewrgross, 2022-02-15:
I did actually! It may or may not be useful to you depending on your function, but it turns out that if you send a docking command and it docks, it goes into passive mode! This is the command:

`rostopic pub -1 dock std_msgs/Empty`

If it's on the dock and you press this, it goes passive, and then if you undock, it becomes active again:

`rostopic pub -1 undock std_msgs/Empty`

I included a bit more info here: https://crashbot.atlassian.net/wiki/spaces/BOT/pages/31293441/Power

and here: https://crashbot.atlassian.net/wiki/spaces/BOT/pages/32014337/State+Overview