AutonomyLab/create_robot #15: Add support for main brush, side brush, and vacuum motors

jacobperron, 2016-03-28:
Add subscribers to listen for the three different motor commands.

zserlin1, 2018-02-26:
Is this still in the works?

jacobperron, 2018-02-26:
@zserlin1 I haven't had time to work on this. There was some work done on this a while ago (#36), perhaps I'll get to it one weekend. In the meantime, PRs always welcome :)

To make the API more clear for developers, I think using a custom message for controlling the motors would be good. A proposal for `ca_msgs/MotorSetpoint`:

        # For the main and side brush motors, provide a duty cycle in the range [-1, 1]
        # The range of acceptable values for the vacuum motor is [0, 1]
        float32 duty_cycle

nnarain, 2021-05-31:
Hi. I was looking into adding a command topic for the vacuum motor. See https://github.com/nnarain/create_robot/commit/d221878e562c30f71ff88a94dc21fc7d98aecd2e

Currently implemented as a `std_msgs/Float32`. Would `create_msgs/MotorSetPoint` be preferable?

vinnnyr, 2021-06-04:
I would prefer a `create_msgs/MotorSetPoint` since [std_msgs is in theory deprecated](https://docs.ros.org/en/foxy/Releases/Release-Foxy-Fitzroy.html#std-msgs)

nnarain, 2021-06-04:
Was not aware of that. Thanks!

jacobperron, 2021-06-10:
Yes, I am also in favor of a semantically meaning message (e.g. like `create_msgs/MotorSetPoint`).

AdronTech, 2022-04-24:
Would it also be possible to merge this into the foxy branch?

jacobperron, 2022-04-26:
Here's a port for Foxy: https://github.com/AutonomyLab/create_robot/pull/97

PR #97 (2022-04-26): Add motor control
Port #91 to Foxy.

Resolves #15
closed 2022-04-26 by jacobperron