AutonomyLab/create_robot #20: Use sensor_msgs/Range for bumpers?

jacobperron, 2016-04-02:
It was suggested that maybe using the [sensor_msgs/Range](http://docs.ros.org/jade/api/sensor_msgs/html/msg/Range.html) message type might be a nice alternative to replace the custom Bumper message. It could be used as follows:  
`field_of_view = 0.34` _(bumper size)_
`max_range = 0` _(no contact)_
`min_range = -0.00891` _(bumper contact)_
Bumper is pressed if: `msg.range < 0`

Unfortunately, this message type would only encapsulate one of the two contact sensors on Create and I don't like the idea of having two separate messages for each bumper.

Also, the light sensors would then have to be moved to a different message type. Perhaps [sensor_msgs/Illuminance](http://docs.ros.org/api/sensor_msgs/html/msg/Illuminance.html)?

betaupsx86, 2017-01-22:
Definitely not. It will only pollute the publisher, subscriber space by having a message per bumper. The custom message works perfectly and is a cleaner solution. However, I suggest the physical and light bumper get their own separate custom messages. Having a custom message follows the Open Interface closer IMO with boolean values and signal strengths. It says nothing about illuminance.

jacobperron, 2017-01-23:
@betaupsx86 Thanks, I think I will implement your suggestion.