AutonomyLab/create_robot #22: Add cliff sensor support

jacobperron, 2016-04-19:
At the very least the driver can publish a Boolean message signalling whether or not a cliff is detected.  
Alternatively, can also publish a custom message similar to the [Bumper](https://github.com/AutonomyLab/create_autonomy/blob/indigo-devel/ca_msgs/msg/Bumper.msg) message, that contains the raw signal strengths as well as the Boolean cliff signals for all four sensors.

betaupsx86, 2017-01-22:
Yep definitely. I added Cliff custom messages while working with the driver. Just let me know if you are open to contributions

jacobperron, 2017-01-23:
@betaupsx86 Definitely open for contributions :)

Dcruise546, 2019-01-23:
@betaupsx86 Is there any update on the cliff custom message? I am trying to use the sensor for an experimental docking mechanism. Any help will be much appreciated!

girvenavery2022, 2022-04-11:
Hi @jacobperron, I just finished this feature and tested it on my create2. I can confirm it works and can open a pr to contribute it!

jacobperron, 2022-04-12:
@girvenavery2022 There is already a PR adding this feature (#93). Sorry, I've just been slow at getting around to testing it.