AutonomyLab/create_robot #23: Kinetic support?

lucbettaieb, 2016-06-03:
Has this package been tested on Kinetic?  I have an iRobot Create 2 that I plan to run from Ubuntu 16.04 (i think..) on a Raspberry Pi 3.  I can test it if you'd like

jacobperron, 2016-06-07:
I haven't tested it myself, on my TODO list along with releasing binaries (probably won't get to it until July though). I don't see any reason why it shouldn't work, let me know how it goes!

lucbettaieb, 2016-06-22:
### Kinetic on Raspberry Pi 3 w/ Ubuntu Mate 16.04
`Linux turtle-pi 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l armv7l armv7l GNU/Linux`

I think I had done the 'bare bones' install so I had to install the following packages:
 - ros-kinetic-xacro
 - ros-kinetic-robot-state-publisher
 - ros-kinetic-joint-state-publisher
 - ros-kinetic-joy

Added myself to dialout, but still had to chmod /dev/ttyUSB0.  Maybe I needed to restart the service?

Couldn't get the `joy` stuff working.  I'll try that more later.  Got `teleop-twist-keyboard` running and was able to drive the robot around.

Seems like it's working for now.  Just need to make sure the `joy` stuff works.

jacobperron, 2016-06-22:
After adding yourself to dialout, you need to logout and login for the change to take affect.  
What joystick did you try to use? The default settings work for the Xbox 360 wired controller.

lucbettaieb, 2016-06-22:
I was using a wireless controller on my Desktop with a my master pointed at the robot.  It was probably something wrong with my setup.  I'll try again later and report back.

andre-nguyen, 2016-10-08:
I have tested the package on Kinetic on my desktop. The teleoperation using an xbox 360 controller works.

owner01, 2016-12-13:
Works like a charm on kinetic didnt test all functionality though, odom cmd_Vel and bumper works

clydemcqueen, 2017-03-28:
I'm using it on Kinetic w/ a Create2 for a while now and it's been working well.

closed 2017-12-18 by jacobperron