AutonomyLab/create_robot #24: Error!

AyanGithub, 2016-06-15:
Hello! I want to run my iRobot Create 2 with my Ubuntu Computer that is running ROS Kinetic 16.04 (with maybe a Raspberry Pi 3 in the future) but have had very bad luck. But when I came across AutonomyLab, I gave it a shot. And it kinda worked, but I got this error:
` launch node of type [ca_driver/ca_driver]: can't locate node [ca_driver] in package [ca_driver]`
I don't know why it's doing that, but any help is appreciated. Thanks!

lucbettaieb, 2016-06-15:
After you `catkin_make`, do you see an executable inside ~/your_catkin_ws/devel/lib/ca_driver?

jacobperron, 2016-06-16:
Were there any errors compiling (running `catkin_make` or `catkin build`) ?

AyanGithub, 2016-06-16:
Thanks for the early replies! I don't even see ca_driver in my lib folder in my create_ws and catkin_ws folders. I think more than likely there were no errors compiling catkin build.

AyanGithub, 2016-06-16:
I made another catkin_ws but this time I used catkin_make. I now see an executable inside devel/lib/ca_driver! I'm going to try to see if it works.

AyanGithub, 2016-06-16:
Still same error.

AyanGithub, 2016-06-16:
In my new catkin_ws, I get a new error:
```
~/test_ws/src/create_autonomy-indigo-devel/ca_driver/launch$ roslaunch create_2.launch [desc:=true] [publish_tf:=true]
... logging to /home/ayan/.ros/log/634cecb4-33b1-11e6-a51b-20689dceb569/roslaunch-Ayan-PC-17912.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://Ayan-PC:45947/

SUMMARY
========

PARAMETERS
 * /ca_driver/create_1: False
 * /ca_driver/dev: /dev/ttyUSB0
 * /ca_driver/loop_hz: 30
 * /ca_driver/publish_tf: True
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.0

NODES
  /
    ca_driver (ca_driver/ca_driver)
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [17926]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 634cecb4-33b1-11e6-a51b-20689dceb569
process[rosout-1]: started with pid [17939]
started core service [/rosout]
process[ca_driver-2]: started with pid [17949]
process[robot_state_publisher-3]: started with pid [17957]
process[joint_state_publisher-4]: started with pid [17965]
[ INFO] [1466074766.427460029]: [CREATE] Create 2 model selected
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> >'
  what():  open: No such file or directory
[ca_driver-2] process has died [pid 17949, exit code -6, cmd /home/ayan/test_ws/devel/lib/ca_driver/ca_driver __name:=ca_driver __log:=/home/ayan/.ros/log/634cecb4-33b1-11e6-a51b-20689dceb569/ca_driver-2.log].
log file: /home/ayan/.ros/log/634cecb4-33b1-11e6-a51b-20689dceb569/ca_driver-2*.log

```

AyanGithub, 2016-06-16:
I also did post it on http://answers.ros.org/question/236849/how-do-you-use-the-irobot-create-2-a-ros-laptop-running-kinetic-and-a-kinect-v1-together/ a few days ago, but now I know what driver to use.

jacobperron, 2016-06-16:
It looks like it can't open the serial port. Before running the driver:

1. Ensure your computer is connected to the Create with the 7-pin serial to USB cable.
2. Determine the correct serial device.
    - Run `ls /dev` before and after connecting the USB cable to your computer. You should see a new device in the list after connecting the cable (e.g. `ttyUSB0`).
    - Modify the file `create_2.launch` to set the correct device.
3. Make sure you have permissions to access the serial port.
    - If you already haven't run the command before: `sudo usermod -a -G dialout $USER`, then logout and login again.

AyanGithub, 2016-06-16:
Thanks for the quick reply. I'm so stupid. I didn't plug in the serial cable! After plugging it in, I ran `roslaunch create_2.launch [desc:=true] [publish_tf:=true]` after sourcing which I always forget to do (many of my problems) and the orange LED on the Create Serial Cable is blinking and on the terminal it says,
```
[ INFO] [1466114465.331133969]: [CREATE] Create 2 model selected
[create::Serial] failed to receive data from Create. Check if robot is powered!
[create::Create] retrying to establish serial connection...
[create::Serial] serial error - Operation canceled
[create::Serial] failed to receive data from Create. Check if robot is powered!
[create::Create] retrying to establish serial connection...
[create::Serial] serial error - Operation canceled
[create::Serial] failed to receive data from Create. Check if robot is powered!
[create::Create] retrying to establish serial connection...
[create::Serial] serial error - Operation canceled
[create::Serial] failed to receive data from Create. Check if robot is powered!
[create::Create] failed to connect over serial: timeout
[create::Serial] serial error - Operation canceled
[create::Serial] failed to receive data from Create. Check if robot is powered!
[FATAL] [1466114510.354846035]: [CREATE] Failed to establish serial connection with Create.
[create::Serial] send failed, not connected.

```

But I'm sure after I follow @jacobperron's instructions it should work. Thanks guys!

AyanGithub, 2016-06-16:
Yep, it sure is! Now, I have to run rviz with this... I will try!

AyanGithub, 2016-06-16:
How exactly do you run rviz on this? And also, I can't run the joystick teleop because:
```
ERROR: cannot launch node of type [joy/joy_node]: joy
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/ayan/test_ws/src
ROS path [2]=/home/ayan/turtlebot/src
ROS path [3]=/opt/ros/kinetic/share
ROS path [4]=/opt/ros/kinetic/stacks
ERROR: cannot launch node of type [joy_teleop/joy_teleop.py]: joy_teleop
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/ayan/test_ws/src
ROS path [2]=/home/ayan/turtlebot/src
ROS path [3]=/opt/ros/kinetic/share
ROS path [4]=/opt/ros/kinetic/stacks
```

lucbettaieb, 2016-06-16:
If the node is running, you should be able to type `rosrun rviz rviz` and select the topics you would like to visualize.

AyanGithub, 2016-06-16:
Thanks for quick reply, now it says 
`ERROR: cannot launch node of type [joy_teleop/joy_teleop.py]: joy_teleop`
and for RViz I also tried that before but not much topics were showing. I tried again and still same thing. I will try to take a screenshot. But I did `roslaunch create_2.launch [desc:=true] [publish_tf:=true]` which worked but I did not get many topics in RViz. I also don't know how to send move commands to the Create 2, I tried before but got an error but I don't really worry about that right now.

lucbettaieb, 2016-06-16:
Install joy_teleop the same way you installed joy.  Not sure about RViz.  Maybe not everything is supported yet.

Generally if you do not have a ROS package, you can install it by doing `sudo apt-get install ros-<distro>-<name>-<of>-<package>`

AyanGithub, 2016-06-16:
[image]
[image]

lucbettaieb, 2016-06-16:
^Looks as if you're doing everything correctly.  RViz just can't visualize everything.  Try adding a `RobotModel` manually.

AyanGithub, 2016-06-16:
Thanks for the quick replies! Let me try that.

AyanGithub, 2016-06-16:
I get this error for joy teleop:
`E: Unable to locate package ros-kinetic-joy-teleop`

AyanGithub, 2016-06-16:
How do you add a manual robot model? 
[image]
It might have to do with robot_description

lucbettaieb, 2016-06-17:
@AyanGithub I'll try to answer your remaining questions, but this is not the place to ask them.  Please direct all future questions to ROS Answers.
 - wrt joy_teleop, Im not sure.  It looks like its some custom python program.  `joy` has `joy_node` but no `joy_teleop`.  https://github.com/ros-drivers/joystick_drivers/tree/indigo-devel/joy/src
 - Try changing your base frame.  Looks like the robot model is there but the transforms are incorrect.

@jacobperron You may want to close this ticket.

closed 2016-06-17 by jacobperron