AutonomyLab/create_robot #25: Publish JointState for wheel joints

lopsided98, 2016-06-22:
Unless, I'm misunderstanding something, it seems that the driver should publish JointState messages for the wheels. Without these messages, robot_state_publisher cannot generate transforms for the wheels.

lopsided98, 2016-06-23:
I [forked this repo](https://github.com/lopsided98/libcreate) and added basic support for publishing the joint state. I didn't create a PR because I don't consider it to be complete yet, because it odes not actually publish the position or velocity of the wheels, but at least it keeps the wheels from floating at the origin in rviz.

I need to figure out how to derive the wheel position and velocity from the data the Create provides. This seems like it should be easy on the Create 2, because it provides the raw encoder data, but the Create 1 (which I am using) does not so there is more math involved.

lucbettaieb, 2016-06-23:
What does the create 1 provide?  You can always check out http://wiki.ros.org/irobot_create_2_1 or http://wiki.ros.org/create_node or http://wiki.ros.org/create_driver

lopsided98, 2016-06-23:
The Create 1 provides distance and heading deltas, which can be used to get a pretty good estimate of how far each wheel moved. I'm working on exposing this data through libcreate.

I used to use create_node to control my Create, and it publishes joint state messages, but their position and velocity is always zero (just like my fork).

lopsided98, 2016-06-23:
libcreate actually calculates the how far each wheel moves , but there appears to be some problems with how it does this on the Create 1 (for one, the left and right distances are always the same, see: https://github.com/AutonomyLab/libcreate/blob/master/src/create.cpp#L107).

lopsided98, 2016-06-23:
My latest changes publish distance and velocity in the JointState messages, but this requires changes to libcreate.

jacobperron, 2016-06-25:
@lopsided98 Thanks for opening this issue and the PR. I'll review and merge next week :)

PR #26 (2016-06-28): Publish JointState messages for the wheels
This PR adds support for publishing JointState messages for the wheels on the Create. This allows robot_state_publisher to generate transforms for the wheels. These messages could also allow other nodes to access the velocities of the wheels.

This fixes #25.

To support these changes, AutonomyLab/libcreate#19 must be merged.
closed 2016-06-28 by jacobperron