AutonomyLab/create_robot #32: Roomba 780 Incorrect Odometry

owner01, 2016-11-22:
Hi when using the roomba 780 the Odometry for rotating is not correct.
It will turn a bit less then 1,125 in rviz when only turning a full circle in real life.
Is there an easy way to change this behavior?
Also how would i make changes to libcreate package and use this package i can not seem to locate it on my computer.
Thanks in advance

jacobperron, 2016-11-22:
If you want to make changes to [libcreate](https://github.com/AutonomyLab/libcreate) you should be able to clone it alongside *create_autonomy* in your catkin workspace and build both together with catkin (I would make sure to clean your workspace first to avoid clashing with the version in *devel*).

Incorrect odometry could be due to a difference in wheel diameter and axle length. Currently [axle length](https://github.com/AutonomyLab/libcreate/blob/master/src/types.cpp#L51) is set to 0.235m for Create 2 and [wheel diameter](https://github.com/AutonomyLab/libcreate/blob/master/include/create/types.h#L74) defaults to 0.078m for all models. This doesn't sound like the culprit though, if you are getting major discrepancies between reported and actual rotation.  

It could be a difference in encoder ticks per revolution, but then I would expect the translation reported by odometry to be significantly off too. If you try driving straight for one meter what is the error?

owner01, 2016-11-23:
Hi when using 568.5 Ticks I was able to drive 1 meter straight forward. When using the default 508.8 tick when driving 1 meter I am actually driving around 80 centimeters. So probably it is an difference in the encoder ticks per revolution.

I came at the value 568.5 by calculating it when rotating 8 rotations to the left and determining its drift. Of course this will not take into account many different possibilities like different surfaces. Also when turning to the right I experienced still a small drift after 8 rotations.

jacobperron, 2016-11-23:
Sounds like you have a solution for now. It is probable that the wheel encoders where changed in the 700 series. I'll try contacting iRobot to see if they can provide any insight. 

> I came at the value 568.5 by calculating it when rotating 8 rotations to the left and determining its drift. Of course this will not take into account many different possibilities like different surfaces. Also when turning to the right I experienced still a small drift after 8 rotations.

This is the nature of wheel encoders. There will inevitably be some drift (especially when the robot is rotating).

jacobperron, 2016-11-28:
@owner01 I contacted iRobot and they claim the encoder ticks are the same on all models. Unfortunately, I don't have access to a Roomba 700 or greater to verify the issue, but hopefully others using that robot will stumble across this thread and report their experience.

owner01, 2016-11-29:
@jacobperron interesting also it makes no sense that my roomba would have a different amount of ticks per revolution since on the site: https://store.irobot.com/default/parts-and-accessories/roomba-accessories/700-series/left-wheel-module/83201.html this wheel works for models from 500 to 700 so it is also not different in the wheel size. Would be nice to know what actually causes the difference then.

owner01, 2017-01-12:
@jacobperron wherre did you actually find the diameter for the wheels? When I am measuring my wheels I am getting 7.0 cm instead of 7.8. This could be due using old wheels but don't think they lost 0.4 cm of rubber on each side.

jacobperron, 2017-01-13:
@owner01 I may have done a measurement myself. In the [OI Spec for Create 2](https://cdn-shop.adafruit.com/datasheets/create_2_Open_Interface_Spec.pdf) it is labelled as 7.2cm.

owner01, 2017-01-13:
Ahh yes I am currently using 7.0 since this is what i measured myself this could be due to using old wheels. Thanks for your response.

closed 2017-01-13 by owner01

balanchine, 2017-10-07:
I also want to make changes with my [Roomba 650 robotic vacuum and Roomba 770](url=http://www.leadsrating.com/robotic-vacuums/robotic-vacuum-comparisons/irobot-roomba-650-vs-770/). Any tips or guidance?

Jonah-Chng, 2018-03-01:
I tried cloning libcreate into my catkin_ws/src workspace, however I cannot catkin_make as libcreate is cmake_minimum_required(VERSION 2.8.12), and not 2.8.3. How did you get past this problem?