AutonomyLab/create_robot #34: Create 1

DARCLab, 2016-12-16:
The instructions say:
Connect computer to Create's 7-pin serial port
-If using Create 1, ensure that nothing is connected to Create's DB-25 port

Just to be clear, is there support for using DB-25 port rather than the 7-pin port? Also, does anyone know one gets theses cables or a work around? The irobot store doesn't sell them anymore.

jacobperron, 2016-12-17:
I have not had the opportunity to try running the driver using the DB-25. I know you can control the robot with the same protocol through either port, so I guess there shouldn't be any problems.

pradhanshrijal, 2018-11-16:
Has anyone had any success with this method? I'm also considering trying to use this method.

Can I just use a DB25 to USB converter use the driver?

pradhanshrijal, 2018-11-18:
The robot is connected successfully through the DB25 port but this error keeps coming up

[image]

jacobperron, 2018-11-18:
@pradhanshrijal When you say the robot is connected successfully, what do you mean? Can you see the robot in the device list?

If you are using the [default config](https://github.com/AutonomyLab/create_autonomy/blob/90e597ea4d85cde1ec32a1d43ea2dd0b4cbf481c/ca_driver/config/default.yaml#L2), then the driver tries to open `/dev/ttyUSB0`.

pradhanshrijal, 2019-01-15:
Hello,

We were able to successfully connect it. The problem was  that in the RaspberryPi 3B the bluetooth was interfering with the Serial Connection. Once you turn the bluetooth off it started working just fine.

Thanks for the help anyways!

GaganBhat, 2020-07-14:
I'm having the same issue, but with the create2.
When I try running `roslaunch ca_driver create_2.launch`, I get the below error. I've made sure to source my devel/setup.bash and have done `sudo usermod -a -G dialout $USER` and restarted. This is running on a raspberry pi 4 connected to a iRobot Roomba Create2 via the normal serial USB cable. I have also ensured that the roomba is connected to the default dev/tty/USB0.
```
setting /run_id to fcd666f0-c5f5-11ea-bf57-dca63263c8e0
process[rosout-1]: started with pid [7613]
started core service [/rosout]
process[ca_driver-2]: started with pid [7617]
[ INFO] [1594747093.564272165]: [CREATE] "CREATE_2" selected
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> >'
  what():  open: No such file or directory
[ca_driver-2] process has died [pid 7617, exit code -6, cmd /home/ubuntu/Programming/create_ws/devel/lib/ca_driver/ca_driver __name:=ca_driver __log:=/home/ubuntu/.ros/log/fcd666f0-c5f5-11ea-bf57-dca63263c8e0/ca_driver-2.log].
```

GaganBhat, 2020-07-14:
@jacobperron 
I'm having the same issue, but with the create2.
When I try running `roslaunch ca_driver create_2.launch`, I get the below error. I've made sure to source my devel/setup.bash and have done `sudo usermod -a -G dialout $USER` and restarted. This is running on a raspberry pi 4 connected to a iRobot Roomba Create2 via the normal serial USB cable. I have also ensured that the roomba is connected to the default dev/tty/USB0.
```
setting /run_id to fcd666f0-c5f5-11ea-bf57-dca63263c8e0
process[rosout-1]: started with pid [7613]
started core service [/rosout]
process[ca_driver-2]: started with pid [7617]
[ INFO] [1594747093.564272165]: [CREATE] "CREATE_2" selected
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> >'
  what():  open: No such file or directory
[ca_driver-2] process has died [pid 7617, exit code -6, cmd /home/ubuntu/Programming/create_ws/devel/lib/ca_driver/ca_driver __name:=ca_driver __log:=/home/ubuntu/.ros/log/fcd666f0-c5f5-11ea-bf57-dca63263c8e0/ca_driver-2.log].
```