AutonomyLab/create_robot #35: Deviation driving straight

owner01, 2017-01-05:
I am currently experiencing a problem where when I want to drive straight forward there will be a deviation to the right.
The odometry will also show this deviation. 
I can simply solve it sending less power to the right wheel. Can I use the straight_radius parameter to change this behaviour or is there any other way then chaging create.cpp?

owner01, 2017-01-05:
So apparantly it was the brush.
So now I removed the brush and the roomba thinks it is driving exactly straight forward but it is having a small deviation to the left (aroudn 5 degrees).
I tried printing the encode data by making changes create.cpp in libcreate but unfortunately it does not seem to have any effect on the system. Even when removing functions they still work.

jacobperron, 2017-01-05:
You can experiment by changing [this line](https://github.com/AutonomyLab/create_autonomy/blob/indigo-devel/ca_driver/src/create_driver.cpp#L135) to use a different API call from libcreate ([driveRadius](https://github.com/AutonomyLab/libcreate/blob/master/include/create/create.h#L184) or driveWheels).
 
I've had a similar experience with Create 2, but I'm not convinced it's a problem with libcreate or create_autonomy. I guess it might just be a problem with the motor controller onboard the robot.

owner01, 2017-01-12:
Its probably due to my roomba. One wheel has considerable more resistance then the other

closed 2017-01-12 by owner01