AutonomyLab/create_robot #41: Create 1 Odometry Not Changing

bellz867, 2017-07-10:
Hello,
We have just started using the create_autonomy package on our create 1 roombas but the odom message seems to never update. The odom message continuously outputs a unit quaternion for orientation and zeros for the position, linear velocity, and angular velocity after pushing the robot around by hand. We have the configuration as default with the "robot_model:" parameters set to CREATE_1. Is there a configuration problem or does the odometry not work on the create 1s?
Thank you for your time.

jacobperron, 2017-07-12:
I recall that last time I checked the odometry messages coming from one of our Create 1's that it was pretty terrible, although it was updating. It is a [known issue](http://answers.ros.org/question/31935/createroomba-odometry) that Create 1 odometry is very unusable. I think the quality may vary depending on firmware version (unfortunately updating firmware is no longer supported by iRobot and I'm unaware of a method to do so). Your best bet may be to simple integrate velocity commands over time as an estimate of odometry.

bellz867, 2017-07-21:
Thank you for responding. On which message are the velocities published? I thought just the odom topic had them.

jacobperron, 2017-07-21:
Apologies, I missed the part about pushing the robot around by hand. Can you check if the odometry updates if you send velocity commands to the driver? It may be the case that the firmware will not update odometry if no drive commands are received.

bellz867, 2017-07-25:
The odometry still doesn't change but the joint states are. Thank you for your help.

closed 2017-07-25 by bellz867

stephen-tang, 2017-08-07:
@bellz867 Hi, I am facing the same problem, so have you found any driver for Create 1 that publishes proper odometry messages?