AutonomyLab/create_robot #42: Odom is wrong on Create 1

mzahana, 2017-11-20:
Hi.
I am working with Create 1 and no matter how the robot moves the only change in `/odom` is in `pose.position.x` only! all others are zeros. Also, the orientation is always `x=0, y=0, z=0, w=1` 

I am building the package from source on ROS Kinetic/ Ubuntu 16

Here is an output of the `odom` topic, after i did some random moves:
```bash
header: 
  seq: 6659
  stamp: 
    secs: 1511204611
    nsecs: 762806633
  frame_id: odom
child_frame_id: base_footprint
pose: 
  pose: 
    position: 
      x: 1.17099511623
      y: 0.0
      z: 0.0
    orientation: 
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
  covariance: [0.6134979128837585, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 18.377765655517578, 0.0, 0.0, 0.0, 22.565868377685547, 0.0, 0.0, 1.7976931348623157e+308, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.7976931348623157e+308, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.7976931348623157e+308, 0.0, 0.0, 22.565868377685547, 0.0, 0.0, 0.0, 36.8668212890625]
twist: 
  twist: 
    linear: 
      x: 0.0
      y: 0.0
      z: 0.0
    angular: 
      x: 0.0
      y: 0.0
      z: 0.0
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.7976931348623157e+308, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.7976931348623157e+308, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.7976931348623157e+308, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```

jacobperron, 2017-11-21:
This has been reported previously #41. I recall experiencing the same issue with Create 1 previously. I can take another look into the code, but I'm pretty sure it is an error in the firmware.

I've been considering adding an option to the driver so that it will report integrated velocity commands as "odometry" (similar to the [autolab-rapi](https://github.com/AutonomyLab/autolab-rapi/blob/master/librapichatterbox/cbdriver.cpp#L411) driver).

mzahana, 2017-11-22:
@jacobperron Thanks for your reply. Is upgrading the Create 1 firmware possible? If yes, is there a newer firmware that fixes this problem? Thanks.

jacobperron, 2017-11-22:
@mzahana iRobot no longer supports firmware upgrades for Create 1. I am not aware of a way to do so.

I've just pushed an experimental change to the branch [feat-odom-estimate](https://github.com/AutonomyLab/create_autonomy/tree/feat-odom-estimate). I have not tested it yet and won't have access to a Create 1 until next week, but you're welcome to give it a try. The idea is to integrate the requested velocity commands to produce the odometry estimate. This means that pushing the robot around by hand with have no affect on odometry, but driving it around should produce a better estimate than the firmware.

closed 2018-10-18 by jacobperron