AutonomyLab/create_robot #43: Need help for create_autonomy

htrinh96, 2018-01-21:
I'm doing a localization project on iRobot Create 2 using this create_autonomy package. 
I have some questions:

1. What are the units for the positions x, y and velocities? - I moved the robot in a measured distance (50 cm = 1.64 ft) and my x position was 1.4. The only plausible unit I can see is ft, but I'm not 100% sure. For the velocity, I think the unit is m/s. However, I need someone to confirm on this. 

2. If unit of position is indeed ft, how can I change it to meters ? Since the unit of wheelDiameter is meter so I think might as well just keep everything consistent.

3. I already changed the wheelDiameter to 0.071m according to my physical measurement on the wheel. Just wondering if you guys would suggest changing the number of ticks per revolution (ticks_per_rev) to improve accuracy. If so, what would be the best value for Create 2 (current default value is 508.8).

Thanks. I would really appreciate any help.

jacobperron, 2018-01-24:
1. Units are supposed to be meters for position and m/s for velocities.
2. N/A
3. You could also try adjusting the number of ticks. See #32 for a related discussion.

htrinh96, 2018-01-24:
Thank you for your help. I will try messing around with it.

closed 2018-08-29 by jacobperron

SKT-Peanut, 2019-06-21:
@htrinh96 So what did you end up changing?