AutonomyLab/create_robot #50: catkin build fails

frankjoshua, 2018-05-16:
I'm getting errors about missing header files when I run catkin build.
ca_msgs/ChargingState.h is the first missing file.
The first bad commit is ea79707168690b0ad857271151b39176539ef082
The last commit that builds is 589a15e2978fab41e5827c9f751da56ed6ef2802
I can give more info if you need it but it just looks like a "git add ca_msgs/." was forgotten.

frankjoshua, 2018-05-17:
I'm building in a Docker container. Here is the Dockerfile is used
If I use
git reset --hard 68385b89c673782d7d0931beca66cb21bce5b3bf
after cloning the repo everything works fine.

`
FROM ros:indigo-ros-base
ENV DEBIAN_FRONTEND noninteractive

RUN apt-get update && apt-get install -y \
    g++ \
    vim \
    && rm -rf /var/lib/apt/lists/

RUN bin/bash -c "echo 'source /opt/ros/indigo/setup.bash' >> .bashrc"

RUN /bin/bash -c "source /opt/ros/indigo/setup.bash && \
                  mkdir -p /catkin_ws/src && \
                  cd /catkin_ws/src && \
                  git clone https://github.com/AutonomyLab/create_autonomy.git && \
                  cd create_autonomy && \
                  catkin_init_workspace && \
                  cd /catkin_ws/ && \
                  apt-get update && \
                  rosdep update && \
                  rosdep install --from-paths src -i -y &&\
                  catkin_make && \
                  echo 'source /catkin_ws/devel/setup.bash' >> /.bashrc && \
                  rm -rf /var/lib/apt/lists/"

RUN bin/bash -c "source .bashrc"`

jacobperron, 2018-05-17:
Thanks for reporting the issue. I had been using catkin tools, which hid the dependency bug. I've just updated the main branch, hopefully the issue is resolved now.

frankjoshua, 2018-05-19:
Works great now, thanks for the quick fix!

closed 2018-05-19 by frankjoshua