AutonomyLab/create_robot #54: I'm getting "failed to receive data from Create. Check if robot is powered!" error

Dhiraj100892, 2018-10-11:
Hi
I'm using create2 robot. To start the robot I use create_2.launch from ca_driver. However, I always get  "failed to receive data from Create. Check if robot is powered!" error. I also tried to switch on the robot by pressing the middle button, but it didn't work. 
However, if I remove the battery from the base and insert it back, it magically starts working. But after sometime again it stops working. Is anyone facing the similar issue ?
Is there something I'm doing wrong? 

Thanks for your help

pradhanshrijal, 2018-11-13:
I am having the same issue with create 1. I will try the battery trick and get back to you though.

isankad1990, 2018-11-14:
i faced to the same issue several time during my project. the battery was
removed and was fitted to solve this problem

Thank you.
Best Regards,
Isanka Diddeniya

S. I. A. P. Diddeniya

Research Assistant

Department of Physics,

Faculty of Applied Sciences,

University of Sri Jayewardenepura,

Gangodawila, Nugegoda,

Sri Lanka.

B.Sc. Honors (USJP)

Tell : +94712869344, +94767155757

pradhanshrijal, 2018-11-16:
Still now working for me. I am using Ubuntu 16.04 on a RaspberryPi 3B and ROS Kinetic. Could that be a problem? 

I have tried to connect it with a PC with the same version of ROS and Linux but to no avail.

Can it be anything to do with the baudrate the systems are suppose to communicate in?

jacobperron, 2018-11-16:
> Can it be anything to do with the baudrate the systems are suppose to communicate in?

This could be the culprit. You could try changing the baud argument for the driver or on the Create itself (see the [manual](https://cdn-shop.adafruit.com/datasheets/create_2_Open_Interface_Spec.pdf) for instructions). The Create communicates with 57600 or 19200 (default) baud.

chaosbuster, 2018-12-03:
For me all it took was to uncomment the baudrate parameter line in the xml. Left everything else 'as is', even the baud rate.  I'm using a Roomba 560 running on a Pi 3.

calderpg, 2018-12-26:
I have encountered a similar issue on 18.04 + ROS Melodic where the driver would fail to connect with `serial error - Operation cancelled`.

The solution in this case was a fix in `libcreate`:

In `libcreate/include/create/serial.hpp`:

`#include <boost/enable_shared_from_this.hpp>`
`class Serial : public boost::enable_shared_from_this<Serial> {`

and in `libcreate/src/serial.cpp`, replace all uses of
```
boost::bind(&Serial:onData, this, _1, _2)
```
with
```
boost::bind(&Serial::onData, shared_from_this(), _1, _2)
```

jacobperron, 2019-03-12:
I am unable to reproduce this issue with a Create 1 platform on Ubuntu 18.04, but I would be willing to accept autonomylab/libcreate#38 if others on this thread can confirm it fixes the issue for them.

jacobperron, 2021-09-29:
I'm going to close this with https://github.com/AutonomyLab/libcreate/pull/38 merged. Let me know if there's still an issue.

closed 2021-09-29 by jacobperron