AutonomyLab/create_robot #56: Issue with catkin build

imslros, 2018-12-17:
Hey guys

I´m having a problem with building in my project. 
I work with an iRobot Create 1 and a Raspberry Pi with Raspbian. 
Installing ROS Melodic worked without any issues and now i want to install the package from create_autonomy but the command "catkin build" fails at the point where it comes to the ca_driver and libcreate.

Here is the Error Message:
------
Errors       << ca_driver:cmake /home/pi/ros_catkin_ws/logs/ca_driver/build.cmake.001.log
CMake Error at /home/pi/ros_catkin_ws/src/create_autonomy/ca_driver/CMakeLists.txt:4 (find_package):
  By not providing "Findlibcreate.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by 
  "libcreate", but CMake did not find one.

  Could not find a package configuration file provided by "libcreate" with
  any of the following names:

     libcreateConfig.cmake
     libcreate-config.cmake

  Add the installation prefix of "libcreate" to CMAKE_PREFIX_PATH or set
  "libcreate_DIR" to a directory containing one of the above files. IF
  "libcreate" provides a seperate development package od SDK, be sure it has
   been installed.

--------

I know it already says what i can do to solve this problem... but i really don´t understand it or know  how to do it exactly.
Can any of you help me please?

jacobperron, 2018-12-17:
This is because the dependency *libcreate* is not (yet) released for Melodic. In the mean time, you should be able to clone [libcreate](http://github.com/autonomylab/libcreate) into your workspace alongside create_autonomy and build everything from source. Let me know if this works for you.

imslros, 2019-01-10:
Hey jacob,

Sorry for replying so late. I have just implemented your proposal and it actually works! Thanks a lot for this!

Now I still have the **problem** that I am informed that _no data is received from the Create_. even though there is a connection between the PI and the Create and the Create is also powered up.

Do you have a solution for this problem, too?

Thank you in advance!

jacobperron, 2019-01-11:
> Now I still have the problem that I am informed that no data is received from the Create. even though there is a connection between the PI and the Create and the Create is also powered up.

@imslros Sounds like someone has a potential solution here: https://github.com/AutonomyLab/create_autonomy/issues/54#issuecomment-450024493

Let's keep any further discussion in that thread.

closed 2019-01-11 by jacobperron