AutonomyLab/create_robot #63: Is there a way to connect several roombas at the same time?

epsilon11101, 2019-10-29:
I'm trying to do it, but I don't know what the way is, I know that the issues are not for this, but I don't find an answer on the other hand

jacobperron, 2019-10-30:
From one computer, you should be able to launch multiple instances of `ca_driver` for each robot.
For example, after connecting the robots to your computer, you should be able to see each in the devices list:

```sh
$ ls /dev/ttyUSB*
/dev/ttyUSB0
/dev/ttyUSB1
```

Once you've identified the device paths, you can create a custom parameter file for each robot, changing the `dev` parameter (see the [default.yaml](https://github.com/AutonomyLab/create_autonomy/blob/90e597ea4d85cde1ec32a1d43ea2dd0b4cbf481c/ca_driver/config/default.yaml) for an example).  You should probably update the `base_frame` and `odom_frame` values for each robot so that they don't clash. Then you can load the parameters in a launch file (see [example launch file](https://github.com/AutonomyLab/create_autonomy/blob/90e597ea4d85cde1ec32a1d43ea2dd0b4cbf481c/ca_driver/launch/create_2.launch).

You should also namespace the driver node for each robot so they do not clash. You can do this by adding a `ns` attribute to the node tag in your launch file.

epsilon11101, 2019-10-30:
thanks for the fast response :) , i have one more question , what's a ns attribute?

epsilon11101, 2019-10-31:
i solve it with ns! (namespace) thanks a lot! <3

jacobperron, 2019-10-31:
happy to help!

closed 2019-10-31 by jacobperron