AutonomyLab/create_robot #66: Launching create_autonomy in gazebo?

user98765467, 2020-02-11:
Is it possible to launch create_autonomy to control a robot in gazebo? If so, how?

eborghi10, 2020-02-11:
Hi @user98765467 I'm working on a fork of this repo that works for both the real robot and Gazebo. If you're interested, here is the link: [https://github.com/RoboticaUtnFrba/create_autonomy](https://github.com/RoboticaUtnFrba/create_autonomy)

user98765467, 2020-02-11:
Thank you! I think this is exactly what i was looking for. I got it installed. How do I actually launch this properly for a gazebo robot? I'm totally new to this stuff and I know I'm supposed to type different things into different terminal windows.

eborghi10, 2020-02-11:
You have the wiki with a lot of things explained or the README files under
each directory: https://github.com/RoboticaUtnFrba/create_autonomy/wiki

user98765467, 2020-02-12:
I've tried a bunch of things to the effect of this but just got errors:
cd ~/create_ws/src/create_autonomy/ca_gazebo/launch
roslaunch create_house.launch

```
hannah@hannah-Lenovo-ideapad-100-15IBD:~$ cd ~/create_ws/src/create_autonomy/ca_gazebo/launch
hannah@hannah-Lenovo-ideapad-100-15IBD:~/create_ws/src/create_autonomy/ca_gazebo/launch$ roslaunch create_house.launch
... logging to /home/hannah/.ros/log/6b3211b6-4d48-11ea-b99a-48e2442c20fd/roslaunch-hannah-Lenovo-ideapad-100-15IBD-18035.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 306, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 749, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 721, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 685, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 589, in _include_tag
    inc_filename = self.resolve_args(tag.attributes['file'].value, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 370, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/home/hannah/.local/lib/python2.7/site-packages/rospkg/rospack.py", line 203, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: ca_gazebo
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/opt/ros/kinetic/share

```

Interaze, 2020-11-28:
It is, here's an example launch file for GAZEBO, not the create. You can also include the create file, but this was for debugging
`<launch>
  
  
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" value="$(find simulation_gazebo)/worlds/YOUR.world"/>
    
    <arg name="paused" value="false"/>
    <arg name="use_sim_time" value="true"/>
    <arg name="gui" value="true"/>
    <arg name="recording" value="false"/>
    <arg name="debug" value="false"/>
  </include>

  
  <group ns="Bot_1">
    <param name="robot_description" command="$(find xacro)/xacro.py '$(find create_description)/urdf/create_2.urdf.xacro'" />
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" output="screen" />
    <node name="spawn_urdf" pkg="gazebo_ros" type="spawn_model" args="-param robot_description -urdf -model robot" />
  </group>
</launch>`

eborghi10, 2020-11-30:
Hey, @Interaze  can you open a ticket directly [here](https://github.com/RoboticaUtnFrba/create_autonomy/issues/new/choose)?

I recommend you to use ROS Melodic instead of Kinetic.

The simplest world is [this one](https://github.com/RoboticaUtnFrba/create_autonomy/blob/kinetic-devel/ca_gazebo/launch/create_empty_world.launch), but all spawn the iRobot Create. If you want to not spawn a robot use `NUM_ROBOTS=0`.

Although, if you want to run Gazebo without a Create, just open `roslaunch gazebo_ros empty_world.launch`.

Interaze, 2020-11-30:
I am using melodic, and I wasn't opening a ticket, just showing an answer and example to the question provided. I did however open a ticket for using the IR sensors in gazebo as I do not see it listed in the topics.

Interaze, 2020-11-30:
To extend on this, here is a more organized launch file example, which allows for a custom world to be included for a gazebo launch file.

```
<launch>
  <include file="$(find gazebo_ros)/launch/empty_world.launch">
    <arg name="world_name" value="$(find simulation_gazebo)/worlds/YOUR.world"/>
    <!-- more default parameters can be changed here -->
    <arg name="paused" value="false"/>
    <arg name="use_sim_time" value="true"/>
    <arg name="gui" value="true"/>
    <arg name="recording" value="false"/>
    <arg name="debug" value="false"/>
  </include>

  <!-- Different namespace so you can add multiple robots -->
  <group ns="Bot_1">
    <include file="$(find create_description)/launch/create_2.launch">
    </include>
        <node name="spawn_urdf" pkg="gazebo_ros" type="spawn_model" args="-param robot_description -urdf -model robot" />
  </group>
</launch>
```

jacobperron, 2020-12-01:
Closing this since I think the question has been answered, and it's not an issue with this repository.
Feel free to continue discussing here.

closed 2020-12-01 by jacobperron