AutonomyLab/create_robot #68: For roomba 527 ca_driver create_1.launch is recommended?

tuberosa, 2020-09-04:
Hi I'm using create_autonomy under the condition below. 

roomba: series 527
raspi 3B
connection: UART ( not via serial cable)
Ubuntu Mate 18.04.2 Bionic Beaver
Ros melodic
(Since I'm not pro, information below may be insufficient or may contain useless information.)

According to README, it's written that roomba series 500 user should use ca_driver create_1.launch, but it didn't work for me.
 (Error `failed to establish connection` was shown)

Instead, when I used **create_2.launch** it worked and I was able to rotate roomba using `/cmd_vel`.

Hope this would help someone who is struggling.

tuberosa, 2020-09-05:
What I did worked once, but it started to show error below.
Right after the battery of roomba was shown, serial error occurred and I could not publish rostopic.

Using create_1.launch showed almost the same result except battery info wasn't shown in this case.

Any help would be appreciated.

```
# roslaunch ca_driver create_1.launch
...
SUMMARY
========

PARAMETERS
 * /ca_driver/base_frame: base_footprint
 * /ca_driver/dev: /dev/ttyS0
 * /ca_driver/latch_cmd_duration: 0.2
 * /ca_driver/loop_hz: 10.0
 * /ca_driver/odom_frame: odom
 * /ca_driver/publish_tf: True
 * /ca_driver/robot_model: CREATE_1
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.7

NODES
  /
    ca_driver (ca_driver/ca_driver)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [2118]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 0c540a56-ef27-11ea-b73f-b827eb1057d0
process[rosout-1]: started with pid [2129]
started core service [/rosout]
process[ca_driver-2]: started with pid [2135]
process[robot_state_publisher-3]: started with pid [2137]
[ INFO] [1599276163.997329640]: [CREATE] "CREATE_1" selected
[create::Serial] serial error - End of file
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> >'
  what():  write: Input/output error
[ca_driver-2] process has died [pid 2135, exit code -6, cmd /home/username/create_ws/devel/lib/ca_driver/ca_driver __name:=ca_driver __log:=/home/usename/.ros/log/0c540a56-ef27-11ea-b73f-b827eb1057d0/ca_driver-2.log].
log file: /home/username/.ros/log/0c540a56-ef27-11ea-b73f-b827eb1057d0/ca_driver-2*.log
```

eborghi10, 2020-09-07:
Try rebooting the robot. For a Roomba 600, you have to press SPOT and CLEAN simultaneously.
If not, remove the battery and put it back again.