AutonomyLab/create_robot #76: Accessing IR Omni publisher with the create_2

Interaze, 2020-11-28:
Put simply, it lists that the ir sensors are supported with this repo, but I'm unable to see it as a topic in ros, is this feature only for runtime instead of gazebo?

jacobperron, 2020-12-01:
Only the omni IR sensor is supported, and this referring to the real robot driver. Note, this is a passive sensor (not a range finder). To get an equivalent sensor in a Gazebo simulation, we should modify the [simulation description](https://github.com/AutonomyLab/create_robot/blob/melodic/create_description/urdf/create_base_gazebo.urdf.xacro) with an appropriate plugin. I'm not sure if there exists a plugin for this type of fiducial sensor.

Interaze, 2020-12-05:
Thankyou for the reply, would it be possible to get help to work on this? I'm willing to help out as much as I can, but I have little experience with developing urdf or sdf models.

jacobperron, 2020-12-07:
@Interaze I don't have much time to dedicate, but it does look like gazebo_ros supports [bumper sensors](http://gazebosim.org/tutorials?tut=ros_gzplugins#Bumper). You could try adapting that example, adding it to the [create gazebo description file](https://github.com/AutonomyLab/create_robot/blob/melodic/create_description/urdf/create_base_gazebo.urdf.xacro).