AutonomyLab/create_robot #85: Create2 wheel control is wrong when low speed

gjjab, 2021-02-19:
This picture is my /odom data, I pub topic cmd_vel line.x = 0.03 at first then it turn, but about timestamp is 90, I pub  topic cmd_vel line.x = 0.06, then it go straight perfect.
I think the joint or task space controller maybe have bugs!?
Can any one help!? thanks a lot

[image]

jacobperron, 2021-03-11:
I don't quite understand the issue. Are you saying that when given a small forward velocity that the robot is turning instead of driving straight? Judging by your plots, it appears you are commanding the robot to turn (`twist.twist.angular.z` is angular velocity). To me, the odometry plot seems to correlate with the velocity plot correctly.

Bscout2011, 2021-07-16:
Hi @jacobperron,

I have similar data showing a negative bias with the odometry.

[image]
 The plot above shows the reference commanded velocity is 0.5 m/s. The `/odom` twist linear x velocity is the orange line at about 0.35 m/s. The distance is calculated by the Euclidean distance for the `/odom` (x, y) position.

[image]

The plot above shows a commanded velocity of 0.35m/s, and the `/odom` topic says the actual velocity is about 0.22 m/s.

jacobperron, 2021-07-22:
I'm not convinced the odometry error is caused by the driver.

The wheel encoders on the Create platform will not give very good odometry. I would not be surprised to see systematic odometry error vary significantly from one robot to the next.