AutonomyLab/create_robot #104: Roomba 800 - where to start ?

Gredin67, 2023-05-18:
Is there some demo or tutorial to understand the capabilities of a roomba 896 with ROS1/2 ? @brownem
Is it possible to e.g. run the navigation stack on it ? 

If not which vacuum cleaning robot would be currently the best option to both clean and play with ROS ?

jacobperron, 2023-05-18:
I'm not aware of any demos.

Also, I'm not sure the Roomba has sufficient sensors to run something like the navigation stack. You'll likely want to add a range sensor like a LIDAR or 3D camera or array of sonars. These extra sensors should be driven by a separate computer (like the create_robot driver) and you'd have to figure out how to mount everything to the robot.

In the past, I've toyed around with a battery-powered Raspberry Pi to control the Create platform (e.g. to remote control it or drive it around in simple patterns), but I never attempted running the navigation stack.

I'm going to close this ticket since it is more of a general question, not specifically about the create_robot driver. You might have luck asking your question on [ROS Answers](https://answers.ros.org/questions/) or on [Discourse](https://discourse.ros.org/).

jacobperron, 2023-05-18:
If you happen to have a Roomba 800, you can try connecting a computer to it's serial port and run this driver. If it's successful, you should see a list of ROS topics that you can interact with to drive it around and get sensor values (e.g. the bumpers).

closed 2023-05-18 by jacobperron