AutonomyLab/create_robot #108: Support for ROS 2 Jazzy

ARLunan, 2024-12-07:
Is this release of Debian install to amd64 and arm64 machines planned? Building create_robot throws an error 2 doing a colcon build of libcreate on ubuntu 24.04/ros2 jazzy.

slgrobotics, 2025-02-04:
I have a version that I use with ROS2 Jazzy, current and updated. Feel free to use it:

https://github.com/slgrobotics/create_robot

slgrobotics, 2025-02-04:
@jacobperron @rtv  - I know, it's been 2+ years since the _Iron_ branch and "_Autonomy Lab at SFU_" link leads nowhere - but wouldn't it be reasonable to create a _Jazzy_ branch now? I have code to contribute, working fine on my Create1 robot. 

See above and https://github.com/slgrobotics/robots_bringup/tree/main/Docs/Create1