AutonomyLab/create_robot #109: ROS 2 Jazzy Build on Raspberry Pi 5 Failed

ARLunan, 2024-12-07:
On Raspberry Pi 5/Ubuntu 24.04/ROS 2 machine using the iron branch of this repository and following the build instructions here, the colcon build failed processing the libcreate package. In troubleshooting, on a separate Raspberry Pi /Ubuntu 22.04/ROS 2 Jazzy machine, the build create_robot & libcreate  completed successfully .  So could the necessary code changes be identified or suggested troubleshooting steps to support a UB24/Jazzy build to complete?

Here is the Build error notification:
ubuntu@rp5-ub24j-mb:~/create_autonomy_ws$ colcon build
Starting >>> create_msgs
Starting >>> libcreate
Starting >>> create_description
Finished <<< create_description [0.49s]                                                     
Finished <<< create_msgs [1.64s]                                                             
--- stderr: libcreate                            
In file included from /home/ubuntu/create_autonomy_ws/src/libcreate/src/packet.cpp:3:
/home/ubuntu/create_autonomy_ws/src/libcreate/include/create/packet.h:50:25: error: field ‘info’ has incomplete type ‘const std::string’ {aka ‘const std::__cxx11::basic_string<char>’}
   50 |       const std::string info;
      |                         ^~~~
In file included from /usr/include/c++/13/iosfwd:41,
                 from /usr/include/c++/13/bits/shared_ptr.h:52,
                 from /usr/include/c++/13/memory:80,
                 from /home/ubuntu/create_autonomy_ws/src/libcreate/src/packet.cpp:1:
/usr/include/c++/13/bits/stringfwd.h:72:11: note: declaration of ‘std::string’ {aka ‘class std::__cxx11::basic_string<char>’}
   72 |     class basic_string;
      |           ^~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/create.dir/build.make:146: CMakeFiles/create.dir/src/packet.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
In file included from /home/ubuntu/create_autonomy_ws/src/libcreate/include/create/data.h:39,
                 from /home/ubuntu/create_autonomy_ws/src/libcreate/src/data.cpp:1:
/home/ubuntu/create_autonomy_ws/src/libcreate/include/create/packet.h:50:25: error: field ‘info’ has incomplete type ‘const std::string’ {aka ‘const std::__cxx11::basic_string<char>’}
   50 |       const std::string info;
      |                         ^~~~
In file included from /usr/include/c++/13/iosfwd:41,
                 from /usr/include/c++/13/bits/shared_ptr.h:52,
                 from /usr/include/c++/13/memory:80,
                 from /home/ubuntu/create_autonomy_ws/src/libcreate/include/create/data.h:36:
/usr/include/c++/13/bits/stringfwd.h:72:11: note: declaration of ‘std::string’ {aka ‘class std::__cxx11::basic_string<char>’}
   72 |     class basic_string;
      |           ^~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/create.dir/build.make:132: CMakeFiles/create.dir/src/data.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:111: CMakeFiles/create.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< libcreate [1.97s, exited with code 2]

Summary: 2 packages finished [2.39s]
  1 package failed: libcreate
  1 package had stderr output: libcreate
  3 packages not processed
ubuntu@rp5-ub24j-mb:~/create_autonomy_ws$

slgrobotics, 2025-02-04:
@ARLunan  Take a look at https://github.com/slgrobotics/create_robot

I hope this helps.

slgrobotics, 2025-02-05:
I just compiled these repositories without any problem on my RPi 5 / Ubuntu 24.04 Server, following the guide here:

https://github.com/slgrobotics/robots_bringup/blob/main/Docs/Create1/README.md

git clone https://github.com/slgrobotics/create_robot.git
git clone https://github.com/slgrobotics/libcreate.git